kmackay/micro-ecc #174: Getting error when I build uECC.c with command "armcc --thumb uECC.c"

Ashwinperal, 2020-09-28:
"asm_arm.h", line 425: Error:  #18: expected a ")"
          : [dptr] REG_RW (result), [lptr] REG_RW (left), [rptr] REG_RW (right),
          ^
"asm_arm.h", line 430: Warning:  #12-D: parsing restarts here after previous syntax error
          );
          ^
"asm_arm.h", line 410: Warning:  #177-D: variable "left_word" was declared but never referenced
      uint32_t left_word;
               ^
"asm_arm.h", line 411: Warning:  #177-D: variable "right_word" was declared but never referenced
      uint32_t right_word;

Anybody can help me out to fix this syntax issue with in __asm__ volatile ( .......);  in the asm_arm.inc file for armcc build?

kmackay, 2020-10-07:
Maybe it's not detecting thumb architecture correctly? You could try manually defining uECC_PLATFORM=uECC_arm_thumb (or thumb2, depending on what you're building for) and see if that helps.

xennex22, 2020-12-10:
I was getting this error on the cortex-M using Keil compiler.  This fixed it.

#define uECC_PLATFORM uECC_thumb

yonasc, 2022-01-06:
Hi,
I am getting the same error, I tried uECC_PLATFORM=uECC_arm_thumb, uECC_PLATFORM=uECC_arm_thumb2, none worked. Did anyone solve this. Maybe it is related to the face I am trying to compile in Keil environment?

yonasc, 2022-01-10:
Found the cause of this. The assembly syntax using ':' and then parameters is not supported by Keil 5 compiler

KevinZu, 2022-05-25:
> Found the cause of this. The assembly syntax using ':' and then parameters is not supported by Keil 5 compiler

Which version of Keil will compile successfully,I am getting the same error.