kmackay/micro-ecc #178: Using keys generated from openssl

prakashkrishna, 2020-11-15:
Hi,
I'm able to successfully run the tests in test_ecdsa.c (x86_64, windows). I tried to use the keys generated from openssl instead of uECC_make_key(). But uECC_verify() failed. I generated keys using the following commands:

`openssl ecparam -genkey -name prime256v1 -noout -out ec_private.pem
openssl ec -in ec_private.pem -pubout -out ec_public.pem
openssl ec -in ec_private.pem -noout -text
Private-Key: (256 bit)
priv:
    18:28:19:b6:99:26:8e:38:30:ba:5a:58:f9:26:c0:
    45:0d:88:7d:af:8f:66:5b:c4:00:fb:b4:2d:e5:03:
    be:21
pub:
    04:ab:7d:eb:fb:49:5c:6d:6c:cb:aa:7e:b2:f8:cd:
    0c:b9:54:72:d7:dc:04:4c:ae:c8:a6:08:b0:fc:03:
    f1:ba:3a:05:28:bb:81:b6:fc:5d:47:dc:e7:b1:cc:
    fb:69:21:da:15:c2:5b:44:2f:ff:71:83:f7:6a:48:
    9a:0e:11:65:dd
ASN1 OID: prime256v1
NIST CURVE: P-256

Test code:
#define NUM_ECC_BYTES 32

uint8_t private[NUM_ECC_BYTES] = {
		0x18,0x28,0x19,0xb6,0x99,0x26,0x8e,0x38,0x30,0xba,0x5a,0x58,0xf9,0x26,0xc0,0x45,
		0x0d,0x88,0x7d,0xaf,0x8f,0x66,0x5b,0xc4,0x00,0xfb,0xb4,0x2d,0xe5,0x03,0xbe,0x21
};

uint8_t publicCompressed[2*NUM_ECC_BYTES] = {
		0x04,0xab,0x7d,0xeb,0xfb,0x49,0x5c,0x6d,0x6c,0xcb,0xaa,0x7e,0xb2,0xf8,0xcd,
		0x0c,0xb9,0x54,0x72,0xd7,0xdc,0x04,0x4c,0xae,0xc8,0xa6,0x08,0xb0,0xfc,0x03,
		0xf1,0xba,0x3a,0x05,0x28,0xbb,0x81,0xb6,0xfc,0x5d,0x47,0xdc,0xe7,0xb1,0xcc,
		0xfb,0x69,0x21,0xda,0x15,0xc2,0x5b,0x44,0x2f,0xff,0x71,0x83,0xf7,0x6a,0x48,
		0x9a,0x0e,0x11,0x65,0xdd
};

uint8_t public[2*NUM_ECC_BYTES];
uint8_t hash[32] = {0};
uint8_t sig[64] = {0};

int test_uECC_secp256r1(void) {
    const struct uECC_Curve_t * curve = uECC_secp256r1();

    uECC_decompress(publicCompressed, public, curve);

    int isValid = uECC_valid_public_key(public, curve);
    if (!isValid) {
    	printf("invalid public key\n");
    	return 1;
    }
    else {
    	printf("valid public key\n");
    }

    memcpy(hash, public, sizeof(hash));

    if (!uECC_sign(private, hash, sizeof(hash), sig, curve)) {
        printf("uECC_sign() failed\n");
        return 1;
    }

    if (!uECC_verify(public, hash, sizeof(hash), sig, curve)) {
        printf("uECC_verify() failed\n");
        return 1;
    }

    printf("uECC_verify() success\n");
    return 0;
}`

Note: If I don't uECC_decompress the public key, uECC_valid_public_key fails. I also tried with out decompressing public key. It still fails. Any help is appreciated.

Thanks a lot

kyleharte, 2021-01-28:
Can you please fix the formatting of your code first? 
use `` marks to specify code blocks. Some of your code is being converted to emojis and it's quite hard to read.
Thanks

prakashkrishna, 2021-01-29:
Hi Kyle, its formatted now. Thanks

kyleharte, 2021-01-30:
uECC does not support the key type prime256v1
Try the key pair
`-----BEGIN PUBLIC KEY-----
MFYwEAYHKoZIzj0CAQYFK4EEAAoDQgAESElFAt9pcSJSBLbAb+GSAJ7mDIHdZhgY
MAThGQvItnRJFcn8eU4Ny9J1ljnROcIMF7OdpMsrMkbvgbUKTzndmQ==
-----END PUBLIC KEY-----`

`-----BEGIN PUBLIC KEY-----
MFYwEAYHKoZIzj0CAQYFK4EEAAoDQgAESElFAt9pcSJSBLbAb+GSAJ7mDIHdZhgY
MAThGQvItnRJFcn8eU4Ny9J1ljnROcIMF7OdpMsrMkbvgbUKTzndmQ==
-----END PUBLIC KEY-----`

prakashkrishna, 2021-01-30:
You've copied public key twice :)
uECC supports secp256r1. And according to https://www.cryptosys.net/pki/eccrypto.html secp256r1 and prime256v1 are same.
I need to sign JWT (JSON Web Tokens) with ECDSA using P-256 and SHA-256 RFC 7518 sec 3.4, defined in OpenSSL as the prime256v1 curve.

kyleharte, 2021-01-30:
I am not too sure on your issue as I don't have a test setup to test it but I think the issue would be with the way the library check if a public key is valid. As the public key you have provided is compressed, the library most likely will not decompress this without you calling the decompress function.

prakashkrishna, 2021-01-30:
The public key validation (after decompress) passes. Its the uECC_verify that fails. If you know any other library (without external dependencies) that can do prime256v1 please let me know. Thanks for looking into the issue.

prakashkrishna, 2021-04-16:
It works for me now. The open ssl prints public key including header (0x04) which means its uncompressed. So while calling uECC_verify, we've to provide public key excluding the header as follows:
`ab:7d:eb:fb:49:5c:6d:6c:cb:aa:7e:b2:f8:cd:0c:b9:54:72:d7:dc:04:4c:ae:c8:a6:08:b0:fc:03:f1:ba:3a:05:28:bb:81:b6:fc:5d:47:dc:e7:b1:cc:fb:69:21:da:15:c2:5b:44:2f:ff:71:83:f7:6a:48:9a:0e:11:65:dd`

closed 2021-04-16 by prakashkrishna

upsampled, 2024-01-12:
Automating the extraction of the uncompressed public key according to @prakashkrishna into a bash script:
```
openssl ec -pubin -in pubkey.pem -noout -text |  grep -v "^\w.*" | tr '\n' ' ' | tr -d ' ' | tr -d ':' | cut -c3-
```
You can add `| xxd -r -p | xxd -i` to have the output be as a C literal byte array.