kmackay/micro-ecc #184: How does it compare to a crypto chip (ATECC608) in Arduino projects

KarlZeilhofer, 2021-06-13:
Hi,

I'm wondering, how this lib compares to a dedicated crypto chip (e.g. ATECC608B) regarding speed, security and flexibility in embedded Arduino projects, e.g. with an ATmega328.

Greets, Karl

sleepdefic1t, 2022-02-08:
Hardware-based cryptography schemes are generally more secure and less flexible than software-based schemes.

The tl;dr is that a crypto chip like the ATECC608B acts more like a black box where you give it something to sign and it returns a signature. Software-based schemes typically use keys directly.

if you're familiar with crypto wallets, it's basically like comparing Ledger and MetaMask. They're both great but have inherent trade-offs between security and convenience.

olimexsmart, 2022-05-19:
Hardware-based solutions are much faster. I will experiment soon with a crypto chip and I will be comparing it to this library.  

I'm expecting at least a 10x improvement for the additional cost to be justified.