kmackay/micro-ecc #205: how to set the RNG function?

ZahidKamil, 2023-02-28:
Hi, I would like to know how to set the RNG function. 
```
#include <stdio.h>
#include <stdlib.h>
#include <time.h>

void generate_random_bytes(unsigned char *buffer, size_t size) {
    // Set the seed value
    srand(time(0));

    // Generate random bytes
    for (size_t i = 0; i < size; i++) {
        buffer[i] = (unsigned char) rand();
    }
}

int main() {
    // Generate 10 random bytes
    size_t size = 10;
    unsigned char buffer[size];

    generate_random_bytes(buffer, size);

    // Output the generated random bytes
    printf("Random bytes: ");
    for (size_t i = 0; i < size; i++) {
        printf("%02x", buffer[i]);
    }
    printf("\n");

    return 0;
}
```
How would I set this code as my RNG function in uECC_set_rng?

uc-mani, 2023-03-16:
I guess you can check the line where the default_RNG function is defined.
#if default_RNG_defined
static uECC_RNG_Function g_rng_function = **&default_RNG;**
#else
static uECC_RNG_Function g_rng_function = 0;
#endif

probably replacing it with your function in **"platform-specific.h"**

yasmineantille, 2023-07-20:
You can just set it in your main function. If `generate_random_bytes` is the random function you want to use do it as follows: 
 
```
int main() {
     uECC_set_rng((uECC_RNG_Function)generate_random_bytes);
     // rest of your code
}
```