kmackay/micro-ecc #206: HASH out value differnet on same file?

InspireSemi, 2023-03-15:
Using :    const struct uECC_Curve_t *curve = uECC_secp256k1();

We are using it this way:
uECC_sign(APPPRIVKEY, sha256_output, sizeof(sha256_output), signature, curve);

This is generating a different signature each time we run the same file through this.
Is this expected?  
I assumed that the signature would be the same for the same binary file?

utzig, 2023-03-15:
> I assumed that the signature would be the same for the same binary file?

It's not unless RFC-6979 is implemented, which is not for `uECC_sign`. The library has a function which does a simplified deterministic signature generation, use `uECC_sign_deterministic`.

InspireSemi, 2023-03-15:
I assume the ecc_verify() api just authenticates a given sha2 against the signature.. and does not need the rgn number b to do this.

uc-mani, 2023-03-16:
I think as long as the signature created using **uECC_sign()** is verified by the **uECC_verify()** function, we are good.

InspireSemi, 2023-03-16:
We plan on using the first api on a PC to sign the binaries and the second api on the device to authenticate so we should be ok.

Marc Karasek
Principal Software Engineer
M: 678.770.3788

***@***.***
www.inspiresemi.com<http://www.cryptocoretech.com/>

THIS MESSAGE IS INTENDED ONLY FOR THE USE OF THE INDIVIDUAL OR ENTITY TO WHICH IT IS ADDRESSED AND MAY CONTAIN INFORMATION THAT IS PRIVILEGED, CONFIDENTIAL AND/OR EXEMPT FROM DISCLOSURE UNDER APPLICABLE LAW.  If the reader of this message is not the intended recipient or agent responsible for delivering the message to the intended recipient, then you are hereby notified that any dissemination or copying of this communication is strictly prohibited.  If you have received this electronic transmission in error, please delete it from your system without copying it and notify the sender by reply e-mail  so that our address record can be corrected.  Thank you.

uc-mani, 2023-03-17:
> We plan on using the first api on a PC to sign the binaries and the second api on the device to authenticate so we should be ok. Marc Karasek Principal Software Engineer M: 678.770.3788 ***@***.*** 

Hey marc, my suggestion is after uECC_sign api. Use the verify uECC api on PC as just an additional step for your signature verification. 
I am following the same approach for Firmware Update.

InspireSemi, 2023-03-17:
Great minds….   Am already doing that to verify the signature after packaging…

Marc Karasek
Principal Software Engineer
M: 678.770.3788

***@***.***
www.inspiresemi.com<http://www.cryptocoretech.com/>

THIS MESSAGE IS INTENDED ONLY FOR THE USE OF THE INDIVIDUAL OR ENTITY TO WHICH IT IS ADDRESSED AND MAY CONTAIN INFORMATION THAT IS PRIVILEGED, CONFIDENTIAL AND/OR EXEMPT FROM DISCLOSURE UNDER APPLICABLE LAW.  If the reader of this message is not the intended recipient or agent responsible for delivering the message to the intended recipient, then you are hereby notified that any dissemination or copying of this communication is strictly prohibited.  If you have received this electronic transmission in error, please delete it from your system without copying it and notify the sender by reply e-mail  so that our address record can be corrected.  Thank you.